INTI-CMNB/KiAuto #4: Run eeschema_do with custom EnvironmentVariables

edsammy, 2020-11-21:
I am interested in using KiAuto with custom EnvironmentVariables (paths) in my `kicad_common` config file. These custom EnvironmentVariables prefix the local libraries I use and with the current scheme of creating a new `kicad_common` config file, eeschema_do complains about missing libraries and after 3 tries times-out and gives up.

I have been running it successfully by commenting out [lines 564-567](https://github.com/INTI-CMNB/KiAuto/blob/b21db51828ffb6df1a90c3f03895f680b4d2262a/src/eeschema_do#L564) but am wondering what the more elegant solution to this may be. Possibly just a replacement of the variables that are desired to be set (`ShowEnvVarWarningDialog=0` and `Editor=/bin/cat`) instead of a complete ignoring of the other variables? 

Is this deterministic change to the `kicad_common` file neccesary? I noticed that it is not done in pcbnew_do.

Thanks!

set-soft, 2020-11-24:
Hi!

I added code to copy the environment variables, please check if it works for you.

I prefer making it as deterministic as possible. I don't know what extra options could be enabled that will make the GUI work in a different way. Like poping some extra dialog.

closed 2020-11-24 by set-soft

edsammy, 2020-11-26:
Works great thanks so much! I am really enjoying your work here and with KiBot and look forward to seeing it grow and make KiCad an amazing CI hardware system. Thanks again!

commit 942b8dc3 (2020-11-24): Now environment variables in kicad_common are preserved

Fixes #4