INTI-CMNB/KiAuto #6: Fit page scaling 1 vs 0 when creating PDF from PCB

johannespfister, 2021-08-31:
When i create a PDF from the PCB, set scaling to 0, the output has the scaling of 1 and isn't fit to the page.
Example command:

        pcbnew_do export --scaling 0 --monochrome --no-title ./test_scale.kicad_pcb . F.Cu

The output is 1:1. However if i set the scaling to 1, the output fits the page. Example command:

        pcbnew_do export --scaling 1 --monochrome --no-title ./test_scale.kicad_pcb . F.Cu

The help text says scaling of 0 should fit the page.

johannespfister, 2021-08-31:
Possible fix:

    --- a/pcbnew_do
    +++ b/pcbnew_do
    @@ -511,11 +511,11 @@ def create_pcbnew_config(cfg):
                 text_file.write('PrintSinglePage=%d\n' % (not cfg.separate))
                 # Scaling
                 if int(cfg.scaling) == 1:
    -                text_file.write('PrintScale=0\n')
    +                text_file.write('PrintScale=1\n')
                 elif cfg.scaling:
                     text_file.write('PrintScale=%3.1f\n' % (cfg.scaling))
                 else:
    -                text_file.write('PrintScale=1\n')
    +                text_file.write('PrintScale=0\n')
                 # List all posible layers, indicating which ones are requested
                 for x in range(0, 50):
                     text_file.write('PlotLayer_%d=%d\n' % (x, int(x in used_layers)))

set-soft, 2021-09-06:
Hi @johannespfister !
Thanks for the fix. Please let me know if the patch is working for you.

closed 2021-09-06 by set-soft

johannespfister, 2021-09-13:
It works for me.

commit eeacca84 (2021-09-06): Fixed pcbnew_do export scaling (0 not fitting the page)
Fixes #6