INTI-CMNB/KiAuto #10: Export images from 3D viewer

mdeweerd, 2021-11-13:
From https://github.com/INTI-CMNB/KiBot/issues/99#issuecomment-968129830

I made a sample script to generate captures from the 3D Viewer.  
It's a start to integrate image generation from the 3D viewer by extending KiAuto/src/pcbnew_do .

The basic export would already be great while allowing configuration of:
- Virtual/SMD/THT component selection (by simulating V,S and T keys);
- Zooming the view (simulate mouse roller click 4);
- Back, front export.

Of course there are more options, and one could imagine even doing a video capture of the model being turned around to get a better feel about the 3D aspects of the design.

This was generated using the sample script:

[image]
[image]

Script to be run as `./gen3DCaptures t1` in the docker VM where t1 is the project name of the demo project: 
[gen3DCaptures.zip](https://github.com/INTI-CMNB/KiBot/files/7532676/gen3DCaptures.zip)

closed 2021-11-17 by set-soft

set-soft, 2021-11-17:
Hi @mdeweerd !

Thanks for the script. I implemented it on KiAuto. So now:

```
pcbnew_do 3d_view --zoom=5 --output_name=top.png  --ray_tracing t1.kicad_pcb .
```

Generates:

[image]

And:
```
pcbnew_do 3d_view --zoom=5 --output_name=bottom.png --view=Z --ray_tracing t1.kicad_pcb .
```

Generates:

[image]

For some reason the regular (no ray tracing) outputs are quite raw:

[image]

Not sure if I'm missing some KiCad option or is related to the X virtual server.

mdeweerd, 2021-11-17:
That's great ;-).  I think you'll get this in KiBot soon ;-).

I generated my 3D Images using setsoft/kicad_auto:latest or setsoft/kicad_auto:dev running on Windows with VcXsrvr .  
As the images are generated/exported the the 3DViewer, I'ld be surprised that it depends on the XServer, but why not.
I think raytracing was on, but I did not set it (I think the reset should have).

Anyway, this is a great addition as the 3D model library is more complete than the 2D model library.  If KiCAD would have an orthogal option, then we'd have a true 2D top view as well.

With the variant filtering we'll get a good realisitic view of what the board looks like.

set-soft, 2021-11-17:
> That's great ;-). I think you'll get this in KiBot soon ;-).

Committed a poorly tested implementation a few seconds ago ;-)

set-soft, 2021-11-17:
> If KiCAD would have an orthogal option, then we'd have a true 2D top view as well.

Take a look at the last patch, works using a very delicated mechanism, but you get:

[image]
[image]

mdeweerd, 2021-11-17:
I had a look at the patch - KiCADs way ?! - I do not even see it on screen.  Not WYSIWYG but something like YGWYDS ?  (You get what you do not see?).
But hey, it seems to be there since 2017.  I never noticed the cube...
I didn't search for it because I thought it did not exist yet.

That's great - the previous 2D implementation is clearly outperformed!  Hard to do better than that!

I'll see iv the selection of these icons can't be improved.

mdeweerd, 2021-11-17:
Well, there's no way to select the toolbar from the keyboard - none of my tricks worked :-(.

Anyway, it works, one more reason to use KiAuto (and KiBot - I am going to test that).

commit 8d492734 (2021-11-17): Added 3D viewer capture

Closes #10