INTI-CMNB/KiAuto #20: KiCad text- and image-injector tools integration - How/Where?

hoijui, 2022-05-22:
Hey @set-soft ! :-)
We were in short contact before. I admire your work and attitude a lot! great to see you are still on it!

## The tools explained

I wrote two small tools for KiCad automation. Each one takes as input:

* a KiCad PCB file
* some additional data, either:
  * text (`KEY=VAUE` pairs)
  * images (pixel-based formats)

and outputs a new KiCad PCB file.

The idea is, to be able to easily inject meta-data onto PCBs automatically (especially in CI).
Such data could be the repos URL, the git commit description, commit date and so on,
either as plain text or encoded in a QR-Code.
To keep it flexible, I decided to Support general image injection, and do the QR-Code generation with an external tool (e.g. `qrencode`).

Here the tools:

* <https://github.com/hoijui/kicad-text-injector> - Rust - current version works with KiCad 5 and 6
* <https://github.com/hoijui/kicad-image-injector> - Python - current version works with KiCad 6; KiCad 5 compatibility exists in a branch

## How to use

Right now, what we do is:

1. Generate out [own docker image](https://github.com/FabCityHamburg/kibot-with-extras) based on `kicad_auto`, adding the two tools.
2. Use it in a [test-project (branch: 'testing')](https://github.com/hoijui/for-science-keyboard/actions/runs/1961598531) together with KiBot:
    1. send the hand-crafted KiCad PCB files from the repo through one of our tools
    2. send the output of that through the other of our tools
    3. send the output of that to KiBot

## Question

As you have a much better overview of the landscape:
Do you have an idea/suggestion of how to better handle the whole thing,
as in.. how and where to build, package, integrate the two tools best?
I'd also be happy to give you write access to the repos, if you want/if that makes it easier for you to handle things.

set-soft, 2022-05-29:
HI @hoijui !

Did you have time to see if the current functionality covers what you need?

set-soft, 2022-08-10:
After more than 2 months without confirmation I assume the implemented functionality is enough

closed 2022-08-10 by set-soft