INTI-CMNB/KiAuto #21: kicad2step_do Error: Still running

dsa-t, 2022-06-01:
Using KiCad 6.0.5 and the STEP export on WSL2, I get:

```
*******************************************************************
******        Statistics on Transfer (Write)                 ******

*******************************************************************
******        Transfer Mode = 0  I.E.  As Is       ******
******        Transferring Shape, ShapeType = 0                      ******
** WorkSession : Sending all data
 Step File Name : $tempfile$.step(263092 ents)  Write  Done

STEP file '[filename].step' created.

ERROR:Still running (kicad2step_do - kicad2step_do:61)
DEBUG:Closing pipe with 22647 (kicad2step_do.kiauto.ui_automation - ui_automation.py:46)
DEBUG:Terminating the session recorder (kicad2step_do.kiauto.ui_automation - ui_automation.py:169)
DEBUG:Closing pipe with 22644 (kicad2step_do.kiauto.ui_automation - ui_automation.py:46)
DEBUG:Error level -1 (kicad2step_do - kicad2step_do:146)
 (kibot - out_step.py:96)

```

This seems to be due to the kicad2step process not being terminated in 0.3 seconds after pressing the OK button.

For me, it starts working when I change the delay at this line to 0.5 seconds without extra system load and 1 second under _some_ extra system load:

https://github.com/INTI-CMNB/KiAuto/blob/7b920e87c5b940f7710760b67c057eb89572cb11/src/kicad2step_do#L58

closed 2022-06-02 by set-soft

set-soft, 2022-06-02:
Hi @dsa-t !
Thanks for reporting. Now KiAuto will wait for a full time-out, retrying the OK every 300 ms.
I released it as 1.6.12, please check if it works for you.

BTW: How complex is to setup a WSL2 to run KiBot? Could you write a small tutorial?

set-soft, 2022-06-03:
For *Configure Global Symbol/Footprint Library Table* I opened #24

set-soft, 2022-06-03:
> ### Schematic editor stuck at dialog about not finding the sheet template

Which operation triggered it? I tried moving `/usr/share/kicad/template/` away and `eeschema` didn't complain.
Which file was missing?

set-soft, 2022-06-03:
Hi @dsa-t !

> ### Increased PCB 3D viewer processing time
> The script starts trying to invoke 3D viewer immediately. it doesn't wait for the pcb to load, causing the 3D viewer invocation to retry after a timeout, and not sooner (while this is done properly in eeschema).

I changed the pcbnew window name regular expression so we only detect the window after the PCB was loaded. Should fix this.

dsa-t, 2022-06-04:
Thanks!

The template error happened in update_xml preflight, later when I added skipping of that dialog, KiBot reported the template error in the next steps.

The template path was actually wrong in a sheet.

commit fb993142 (2022-06-02): kicad2step: Now we wait a full time-out on exit

- We retry the "press OK" every 300 ms
Fixes #21