INTI-CMNB/KiAuto #22: Consider setting LIBGL_ALWAYS_SOFTWARE=1 in pcbnew_do

dsa-t, 2022-06-02:
With KiCad 6.0.5 on WSL2, Windows 10, the 3D viewer crashes upon opening:

```
> DEBUG:Found 3D viewer dialog window (1) (pcbnew_do.kiauto.ui_automation - ui_automation.py:375)
> DEBUG:Window id: 4200327 (pcbnew_do.kiauto.ui_automation - ui_automation.py:380)
> ERROR:Command '['xdotool', 'windowfocus', '--sync', '4200327']' returned non-zero exit status 1. (pcbnew_do.kiauto.ui_automation - ui_automation.py:472)
> DEBUG:Closing pipe with 21714 (pcbnew_do.kiauto.ui_automation - ui_automation.py:46)
> DEBUG:Terminating 21714 (pcbnew_do.kiauto.ui_automation - ui_automation.py:58)
> DEBUG:Terminating the session recorder (pcbnew_do.kiauto.ui_automation - ui_automation.py:169)
> DEBUG:Closing pipe with 21713 (pcbnew_do.kiauto.ui_automation - ui_automation.py:46)
> DEBUG:Terminating 21713 (pcbnew_do.kiauto.ui_automation - ui_automation.py:58)
```

This can be prevented by adding this to pcbnew_do:

```
    logger.debug('Select a filename -> '+cfg.select_a_filename)
    logger.debug('Print -> '+cfg.print_dlg_name)

    # Force english + UTF-8
    os.environ['LANG'] = get_en_locale(logger)
+   # Force Mesa software rendering (otherwise, the 3D viewer may crash)
+   os.environ['LIBGL_ALWAYS_SOFTWARE'] = "1"

    # Make sure the input file exists and has an extension
    check_input_file(cfg, NO_PCB, WRONG_PCB_NAME)
    cfg.board = load_pcb(cfg.input_file)
```

But notice that there may be artifacts like this:

[image]

Notice that there's some antialiasing, which for some reason cannot be disabled. But having "Disable anti-aliasing" "While moving" enabled seem to mitigate these artifacts (but only while moving, of course).

---
For a note, the 3D viewer crashes in the same way when running with a "more visible" X server (VcXsrv), and even if the native OpenGL support is disabled in VcXsrv, glxinfo still reports:

```
NVD3D10: CPU cyclestats are disabled on client virtualization
NVD3D10: CPU cyclestats are disabled on client virtualization
display: 192.168.16.1:0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
...
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Microsoft Corporation (0xffffffff)
    Device: D3D12 (NVIDIA GeForce GTX 950M) (0xffffffff)
    Version: 21.2.6
    Accelerated: yes
    Video memory: 10160MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 3.3
    Max compat profile version: 3.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.0
OpenGL vendor string: Microsoft Corporation
OpenGL renderer string: D3D12 (NVIDIA GeForce GTX 950M)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 21.2.6
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
```

and with LIBGL_ALWAYS_SOFTWARE=1:

```
display: 192.168.16.1:0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
...
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Mesa/X.org (0xffffffff)
    Device: llvmpipe (LLVM 12.0.0, 256 bits) (0xffffffff)
    Version: 21.2.6
    Accelerated: no
    Video memory: 12720MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.5
    Max compat profile version: 3.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
OpenGL vendor string: Mesa/X.org
OpenGL renderer string: llvmpipe (LLVM 12.0.0, 256 bits)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 21.2.6
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
```

set-soft, 2022-06-03:
Hi @dsa-t !
I added a couple of KiCad config settings that should disable AA.
Do you have a small example to reproduce the artifact?

dsa-t, 2022-06-03:
Here's an example of "video" demo project:

[image]

This is viewing pcbnew through x11vnc/Xvfb with LIBGL_ALWAYS_SOFTWARE=1
opengl_AA_mode seem to not actually be able to disable antialiasing.

Does this happen on a normal Linux distro?

set-soft, 2022-06-03:
> Does this happen on a normal Linux distro?

Yes: 
[image]
[image]

But I don't know if this is really related to AA

dsa-t, 2022-06-03:
Does it render normally while moving and "Disable anti-aliasing" "While moving" option is enabled?

set-soft, 2022-06-03:
I don't really care much about it, when using the GUI I use the accelerator and when using KiBot just enable ray tracing (after fine tuning the options), the previous image becomes:

[image]

Also: On a real X server using LIBGL_ALWAYS_SOFTWARE=1 doesn't generate artifacts, is the Virtual X Server the one with problems. On the real X I get:

[image]

dsa-t, 2022-06-03:
Well, here's a theory:

---
https://gitlab.com/kicad/code/kicad/-/blob/6.0/3d-viewer/common_ogl/ogl_attr_list.cpp#L80

Here, in your case, after calling wxGLCanvas::IsDisplaySupported (which may, or may not change the OpenGL state), the antialiasing mode falls back to ANTIALIASING_MODE::AA_NONE.
In my case, this setting doesn't change anything visually, so the mode should be ANTIALIASING_MODE::AA_NONE too.

That clears the setup of [ARB_multisample ](https://www.khronos.org/registry/OpenGL/extensions/ARB/ARB_multisample.txt) variables:
GLX_SAMPLE_BUFFERS_ARB,
GLX_SAMPLES_ARB.

I don't know what defaults the driver defaults them to, but wxGLCanvas::IsDisplaySupported calls failing should indicate that multisampling doesn't work.

---
https://gitlab.com/kicad/code/kicad/-/blob/6.0/3d-viewer/3d_rendering/opengl/render_3d_opengl.cpp#L589
GL_MULTISAMPLE is enabled even when the mode is ANTIALIASING_MODE::AA_NONE:

```
    // Initial setup
    glDepthFunc( GL_LESS );
    glEnable( GL_CULL_FACE );
    glFrontFace( GL_CCW );    // This is the OpenGL default
    glEnable( GL_NORMALIZE ); // This allow OpenGL to normalize the normals after transformations
    glViewport( 0, 0, m_windowSize.x, m_windowSize.y );

    if( m_boardAdapter.GetFlag( FL_RENDER_OPENGL_AA_DISABLE_ON_MOVE ) && aIsMoving )
        glDisable( GL_MULTISAMPLE );
    else
        glEnable( GL_MULTISAMPLE );

    // clear color and depth buffers
    glClearColor( 0.0f, 0.0f, 0.0f, 1.0f );
    glClearDepth( 1.0f );
    glClearStencil( 0x00 );
    glClear( GL_COLOR_BUFFER_BIT | GL_DEPTH_BUFFER_BIT | GL_STENCIL_BUFFER_BIT );
```

It renders correcly for me when `glDisable( GL_MULTISAMPLE );` is being called, that's why I asked about that.

---

The artifacts disappear when using `GALLIUM_DRIVER=softpipe`, though it's extremely slow.

WIth an older libgl1-mesa-dri on a normal Linux distro, it renders correctly in xvfb (with LIBGL_ALWAYS_SOFTWARE=1):
```
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
...
client glx vendor string: Mesa Project and SGI
client glx version string: 1.4
...
GLX version: 1.4
...
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: VMware, Inc. (0xffffffff)
    Device: llvmpipe (LLVM 10.0.0, 256 bits) (0xffffffff)
    Version: 20.0.8
    Accelerated: no
    Video memory: 15908MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 3.3
    Max compat profile version: 3.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: llvmpipe (LLVM 10.0.0, 256 bits)
OpenGL core profile version string: 3.3 (Core Profile) Mesa 20.0.8
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
...
OpenGL version string: 3.1 Mesa 20.0.8
OpenGL shading language version string: 1.40
OpenGL context flags: (none)
...
OpenGL ES profile version string: OpenGL ES 3.1 Mesa 20.0.8
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
```

set-soft, 2022-06-06:
I don't know much about OpenGL. But from what you mention it looks like the 
```
if( m_boardAdapter.GetFlag( FL_RENDER_OPENGL_AA_DISABLE_ON_MOVE ) && aIsMoving )
```
is incomplete, it should add some sort of  *or AA is none*
Is that correct?

dsa-t, 2022-06-06:
Exactly.

Also I filed a bug report to KiCad about the crash: https://gitlab.com/kicad/code/kicad/-/issues/11751

That's strange that if I apply the workaround described there, this D3D12 swrast driver always renders _with_ antialiasing, even if `glDisable( GL_MULTISAMPLE );` is called.

If I disable the GPU in device manager, the llvmpipe renderer is used (the same as with LIBGL_ALWAYS_SOFTWARE=1 and GPU enabled)

set-soft, 2022-06-06:
Perhaps you could ask in the KiCad forum about adding an extra condition to the above mentioned if sentence.

commit e7c9eb23 (2022-06-02): Force Mesa GL to use sofware render.

- Helps on some setups (i.e. WSL2, VcXsrv).
- Usually not needed.

Closes #22