INTI-CMNB/KiAuto #26: Inconsistencies in KiCad 6 DRC

charliebruce, 2022-07-01:
Hi,

I recently put in a PR that changed the DRC parsing on KiCad 6 - however it still didn't fix my key issue, which was that excluded DRC violations were still causing errors. 

I initially thought this was a parsing error, however when I did a deeper dive, I noticed that the reports that KiAuto outputs were subtly different to the ones generated through the UI. I looked at the KiCad 6 source, and found that DRC reports are created in two, subtly different ways. Specifically:

https://gitlab.com/kicad/code/kicad/-/blob/master/pcbnew/dialogs/dialog_drc.cpp outputs:

```
    fprintf( fp, "\n** Found %d DRC violations **\n", count );

    for( int i = 0; i < count; ++i )
    {
        const std::shared_ptr<RC_ITEM>& item = m_markersProvider->GetItem( i );
        SEVERITY severity = item->GetParent()->GetSeverity();

        if( severity == RPT_SEVERITY_EXCLUSION )
            severity = bds.GetSeverity( item->GetErrorCode() );

        fprintf( fp, "%s", TO_UTF8( item->ShowReport( units, severity, itemMap ) ) );
    }
```

However https://gitlab.com/kicad/code/kicad/-/blob/master/pcbnew/python/scripting/pcbnew_scripting_helpers.cpp outputs:

```
    for( const std::shared_ptr<DRC_ITEM>& item : violations )
    {
        SEVERITY severity = item->GetParent()->GetSeverity();
        fprintf( fp, "%s", TO_UTF8( item->ShowReport( aUnits, severity, itemMap ) ) );
    }
```

Note that the report generated by Python scripting doesn't handle exclusions. It looks like this is a known issue on the KiCad side: https://gitlab.com/kicad/code/kicad/-/issues/11562

Until the Python API is consistent and reports all the required info, I'd suggest that KiAuto uses the UI automation method, rather than the Python API.

Are you happy with that as a proposed fix? If so, I can put in a PR.

set-soft, 2022-07-04:
Hi @charliebruce !
I want to avoid using the UI, is really slow and prone to fail.
We could try loading the board.design_settings.drc_exclusions from the project.
I never used it ... do you have an example?

set-soft, 2022-08-10:
Hi @charliebruce !
Now KiAuto has the `-F` option for the `run_drc` command, it forces the use of the GUI. With the use of the interporser this is now quite reliable.
KiBot now has a global option named `drc_exclusions_workaround` to use this flag until the KiCad issue is solved.
I think this problem is solved now.

closed 2022-08-10 by set-soft

charliebruce, 2022-08-10:
Amazing, thanks so much! I'll give this a go.