INTI-CMNB/KiAuto #32: BOM CSV no longer generated as of v2.0.0

0xabe1, 2023-02-05:
README.md says:

> After running it ./YOUR_SCHEMATIC.xml will be updated. You'll also get DESTINATION/YOUR_SCHEMATIC.csv contain a very basic BoM generated using KiCad's bom2grouped_csv.xsl template.

However, the CSV file appears to no longer be generated. I tested v2.1.1 and v2.0.0 and both are missing the CSV. Version 1.6.15 does still generate the CSV.

I don't see any commit messages specifically referencing its removal, so I'm hoping this is just a bug?

Thanks.

set-soft, 2023-02-06:
Hi @0xabe1 !
Thanks for pointing this out.
Is not a bug, the CSV was side effect of generating the XML.
To generate a BoM use [KiBot](https://github.com/INTI-CMNB/KiBot/), you'll have much more control on what is generated.

closed 2023-02-06 by set-soft

0xabe1, 2023-02-06:
Ok, thanks!