INTI-CMNB/KiAuto #33: "Electrical Rules Checker dialog not responding" error when running ERC

LIV2, 2023-10-22:
Hi,

I'm attempting to use KiAuto on Ubuntu Mantic on ARM64
Kicad version 7.0.7 installed from the Ubuntu package repository

I get "ERROR:Electrical Rules Checker dialog not responding" when attempting to run ERC

Attached are the screencast & logs

[run_erc_eeschema_screencast.ogv.zip](https://github.com/INTI-CMNB/KiAuto/files/13064419/run_erc_eeschema_screencast.ogv.zip)

``` 
root@b43436903895:/build/Kicad# /usr/local/bin/eeschema_do -r -vv run_erc -o /build/Kicad/A2000-Coprocessor-Debug-erc.txt -g 50 -f /build/Kicad/kibot_errors.filter /build/Kicad/A2000-Coprocessor-Debug.kicad_sch /build/Kicad/
DEBUG:Detected KiCad v7.0.7 (7.0.7+dfsg-1 7000007) (eeschema_do - misc.py:155)
DEBUG:Config path /root/.config/kicad/7.0 (eeschema_do - misc.py:180)
DEBUG:KiCad environment: {} (eeschema_do - misc.py:300)
DEBUG:KiCad share dir: /usr/share/kicad/ (eeschema_do - misc.py:242)
DEBUG:Defining KICAD7_FOOTPRINT_DIR = /usr/share/kicad/footprints (eeschema_do - misc.py:252)
DEBUG:Setting timeout scale to 1.0 (eeschema_do - eeschema_do:837)
DEBUG:Loading filter errors (eeschema_do.kiauto.file_util - file_util.py:82)
INFO:Loaded 3 error filters from `/build/Kicad/kibot_errors.filter`
DEBUG:English locale: C.utf8 (eeschema_do - misc.py:338)
DEBUG:Eeschema config: /root/.config/kicad/7.0/eeschema.json (eeschema_do.kiauto.file_util - file_util.py:206)
DEBUG:Moving current config to /root/.config/kicad/7.0/eeschema.json.pre_script (eeschema_do.kiauto.file_util - file_util.py:213)
DEBUG:Creating an eeschema config (eeschema_do - eeschema_do:668)
DEBUG:Selecting plot format pdf (4) (eeschema_do - eeschema_do:671)
DEBUG:{"plot": {"format": 4, "color": true, "frame_reference": true, "color_theme": "_builtin_default", "background_color": false, "hpgl_origin": 0, "hpgl_pen_size": 0.19}, "system": {"first_run_shown": true, "never_show_rescue_dialog": true}, "appearance": {"show_sexpr_file_convert_warning": false, "color_theme": "_builtin_default"}, "window": {"size_x": 1366, "size_y": 960, "grid": {"last_size": 7, "user_grid_x": "50 mil", "user_grid_y": "50 mil"}}} (eeschema_do - eeschema_do:692)
DEBUG:KiCad common config: /root/.config/kicad/7.0/kicad_common.json (eeschema_do.kiauto.file_util - file_util.py:206)
DEBUG:Moving current config to /root/.config/kicad/7.0/kicad_common.json.pre_script (eeschema_do.kiauto.file_util - file_util.py:213)
DEBUG:Creating a KiCad common config (eeschema_do.kiauto.file_util - file_util.py:241)
DEBUG:{"environment": {"show_warning_dialog": false}, "graphics": {"cairo_antialiasing_mode": 0, "opengl_antialiasing_mode": 0}, "system": {"editor_name": "/bin/cat"}} (eeschema_do.kiauto.file_util - file_util.py:253)
DEBUG:User hotkeys config: /root/.config/kicad/7.0/user.hotkeys (eeschema_do.kiauto.file_util - file_util.py:206)
DEBUG:Moving current config to /root/.config/kicad/7.0/user.hotkeys.pre_script (eeschema_do.kiauto.file_util - file_util.py:213)
DEBUG:Creating a user hotkeys config (eeschema_do.kiauto.file_util - file_util.py:221)
DEBUG:Output file: /build/Kicad/A2000-Coprocessor-Debug-erc.txt (eeschema_do - eeschema_do:917)
DEBUG:Waiting for virtual X server ... (eeschema_do.kiauto.ui_automation - ui_automation.py:91)
DEBUG:Current DISPLAY is :1707967580 (eeschema_do.kiauto.ui_automation - ui_automation.py:93)
DEBUG:Checking using ['setxkbmap', '-query'] (eeschema_do.kiauto.ui_automation - ui_automation.py:104)
DEBUG:Recording session with: ['recordmydesktop', '--overwrite', '--no-sound', '--no-frame', '--on-the-fly-encoding', '-o', '/build/Kicad/run_erc_eeschema_screencast.ogv'] (eeschema_do.kiauto.ui_automation - ui_automation.py:166)
DEBUG:Starting eeschema (eeschema_do - eeschema_do:949)
DEBUG:EEschema PID: 579 (eeschema_do - eeschema_do:957)
INFO:Waiting for "Main eeschema window" ...
DEBUG:['xdotool', 'search', '--onlyvisible', '--name', '.*'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:['xdotool', 'getwindowname', '1293'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:get_windows 1293  (eeschema_do.kiauto.ui_automation - ui_automation.py:283)
DEBUG:get_windows end of list (eeschema_do.kiauto.ui_automation - ui_automation.py:285)
DEBUG:['xdotool', 'search', '--onlyvisible', '--name', '.*'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:['xdotool', 'getwindowname', '1293'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:get_windows 1293  (eeschema_do.kiauto.ui_automation - ui_automation.py:283)
DEBUG:get_windows end of list (eeschema_do.kiauto.ui_automation - ui_automation.py:285)
DEBUG:['xdotool', 'search', '--onlyvisible', '--name', '.*'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:['xdotool', 'getwindowname', '1293'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:get_windows 1293  (eeschema_do.kiauto.ui_automation - ui_automation.py:283)
DEBUG:get_windows end of list (eeschema_do.kiauto.ui_automation - ui_automation.py:285)
DEBUG:['xdotool', 'search', '--onlyvisible', '--name', '.*'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:['xdotool', 'getwindowname', '1293'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:get_windows 1293  (eeschema_do.kiauto.ui_automation - ui_automation.py:283)
DEBUG:get_windows end of list (eeschema_do.kiauto.ui_automation - ui_automation.py:285)
DEBUG:['xdotool', 'search', '--onlyvisible', '--name', '.*'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:['xdotool', 'getwindowname', '1293'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:get_windows 1293  (eeschema_do.kiauto.ui_automation - ui_automation.py:283)
DEBUG:get_windows end of list (eeschema_do.kiauto.ui_automation - ui_automation.py:285)
DEBUG:['xdotool', 'search', '--onlyvisible', '--name', '.*'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:['xdotool', 'getwindowname', '1293'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:get_windows 1293  (eeschema_do.kiauto.ui_automation - ui_automation.py:283)
DEBUG:get_windows end of list (eeschema_do.kiauto.ui_automation - ui_automation.py:285)
DEBUG:['xdotool', 'search', '--onlyvisible', '--name', '.*'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:['xdotool', 'getwindowname', '1293'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:get_windows 1293  (eeschema_do.kiauto.ui_automation - ui_automation.py:283)
DEBUG:['xdotool', 'getwindowname', '6291478'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
[... 509 lines cut ...]
DEBUG:['xdotool', 'getwindowname', '6291495'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:get_windows 6291495  (eeschema_do.kiauto.ui_automation - ui_automation.py:283)
DEBUG:['xdotool', 'getwindowname', '6293226'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:get_windows 6293226  (eeschema_do.kiauto.ui_automation - ui_automation.py:283)
DEBUG:['xdotool', 'getwindowname', '6298008'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:get_windows 6298008  (eeschema_do.kiauto.ui_automation - ui_automation.py:283)
DEBUG:get_windows end of list (eeschema_do.kiauto.ui_automation - ui_automation.py:285)
DEBUG:Found Electrical Rules Checker dialog window (1) (eeschema_do.kiauto.ui_automation - ui_automation.py:381)
DEBUG:Window id: 6293225 (eeschema_do.kiauto.ui_automation - ui_automation.py:386)
DEBUG:['xdotool', 'windowfocus', '--sync', '6293225'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:Waiting for 6293225 window to get focus... (eeschema_do.kiauto.ui_automation - ui_automation.py:317)
DEBUG:['xdotool', 'getwindowfocus'] (eeschema_do.kiauto.ui_automation - ui_automation.py:218)
DEBUG:Currently focused id: 6293225 (eeschema_do.kiauto.ui_automation - ui_automation.py:320)
ERROR:Electrical Rules Checker dialog not responding (eeschema_do - eeschema_do:469)
DEBUG:Closing pipe with 579 (eeschema_do.kiauto.ui_automation - ui_automation.py:48)
DEBUG:Terminating 579 (eeschema_do.kiauto.ui_automation - ui_automation.py:61)
DEBUG:Terminating the session recorder (eeschema_do.kiauto.ui_automation - ui_automation.py:172)
DEBUG:Closing pipe with 578 (eeschema_do.kiauto.ui_automation - ui_automation.py:48)
DEBUG:Terminating 578 (eeschema_do.kiauto.ui_automation - ui_automation.py:61)
DEBUG:Restoring old eeschema config (eeschema_do.kiauto.file_util - file_util.py:188)
DEBUG:Restoring old KiCad common config (eeschema_do.kiauto.file_util - file_util.py:188)
DEBUG:Restoring old user hotkeys config (eeschema_do.kiauto.file_util - file_util.py:188)
DEBUG:Checking if /build/Kicad/A2000-Coprocessor-Debug.kicad_pro was modified (eeschema_do.kiauto.file_util - file_util.py:338)
DEBUG:Checking if /build/Kicad/A2000-Coprocessor-Debug.kicad_prl was modified (eeschema_do.kiauto.file_util - file_util.py:338)
```

set-soft, 2023-10-22:
Hi @LIV2 !

Which version are you using? Try using the current git code. 
I'll release it as soon as possible, I noticed some changes in KiCad behavior (from 7.0.5 to 7.0.8)

The ERC fails when no interposer is available, which is the case of ARM64. BTW: the interposer mechanism is faster and more reliable. We could try to make it work for ARM64. There are some interest of other people (using docker images on macOS with CPU emulation).

LIV2, 2023-10-23:
Hi @set-soft 

I was using the latest from the Master branch, I've installed v2.2.8 via pip now but still have the same issue

```
root@b43436903895:/build/Kicad# eeschema_do  --version
eeschema_do 2.2.8 - Copyright 2018-2023, INTI/Productize SPRL - License: Apache 2.0
```
[eeschema_do_2.2.8.txt](https://github.com/INTI-CMNB/KiAuto/files/13069846/eeschema_do_2.2.8.txt)

[run_erc_eeschema_screencast.ogv.zip](https://github.com/INTI-CMNB/KiAuto/files/13069818/run_erc_eeschema_screencast.ogv.zip)

LIV2, 2023-10-23:
Hi,

Yep it does indeed work with 7.0.8 (without interposer) thanks for that :)

I built libinterposer.so and renamed it to libinterposer_aarch64.so and found that when I run eeschema_do it mostly works but gets stuck at "Wait for ERC file creation"

[run_erc_eeschema_screencast 2.ogv.zip](https://github.com/INTI-CMNB/KiAuto/files/13079594/run_erc_eeschema_screencast.2.ogv.zip)
[eeschema_do_interposer_aarch64.txt](https://github.com/INTI-CMNB/KiAuto/files/13079596/eeschema_do_interposer_aarch64.txt)

set-soft, 2023-11-08:
Hi @LIV2 !

Sorry for the big delay. The problem seems to be a missing call to close the ERC report file. Can you try defining the environment variable *KIAUTO_USE_PID_FOR_CREATE* to 1? Something like `export KIAUTO_USE_PID_FOR_CREATE=1` before running KiAuto.

It looks like I'm missing some of the close calls. The bizarre thing is that the above workaround was designed to be used when using VirtioFS on an emulated docker image running on macOS! 

I'm catching *fclose* and *close*. But looking at the code on the *open* side I'm catching: *fopen64*, *fopen*, *open64*  and *open*. I think this is because I found that *fclose64* and *close64* doesn't really exist. But perhaps they do on ARM64. I'm adding them to the interposer. You can try recompiling it, define *KIAUTO_USE_PID_FOR_CREATE* to 0 and then check if this is the real problem. The patch is e7d5edde11d757fef6139ae853950edbf9011271

set-soft, 2023-11-24:
Hi @LIV2 !

Any news about it?  

This issue may be related to INTI-CMNB/KiBot#367, INTI-CMNB/KiBot#509 and INTI-CMNB/kicad_auto#14

LIV2, 2024-01-14:
Hi, Sorry for the late response!

I will try to test this out again soon and get back to you

LIV2, 2024-01-30:
Testing in a Debian trixie docker container, Kicad 7.0.10 with the latest kiauto commit (19ec2fa)

With interposer:
```
INFO:Waiting for Eeschema window ...
DEBUG:Waiting Eeschema to start and load the Schematic (eeschema_do - interposer.py:675)
DEBUG:Waiting for `['GTK:Window Title:', 'PANGO:0:']` starts=True times=1 (eeschema_do - interposer.py:151)
DEBUG:Closing pipe with 455 (eeschema_do.kiauto.ui_automation - ui_automation.py:48)
DEBUG:Terminating 455 (eeschema_do.kiauto.ui_automation - ui_automation.py:61)
DEBUG:Terminating the session recorder (eeschema_do.kiauto.ui_automation - ui_automation.py:172)
DEBUG:Closing pipe with 454 (eeschema_do.kiauto.ui_automation - ui_automation.py:48)
DEBUG:Terminating 454 (eeschema_do.kiauto.ui_automation - ui_automation.py:61)
Traceback (most recent call last):
  File "/usr/local/bin/eeschema_do", line 975, in <module>
    can_retry = wait_eeschema_start_by_msg(cfg) if cfg.use_interposer else wait_eeschema_start(cfg)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/bin/eeschema_do", line 772, in wait_eeschema_start_by_msg
    wait_start_by_msg(cfg)
  File "/usr/local/lib/python3.11/dist-packages/kiauto/interposer.py", line 684, in wait_start_by_msg
    res = wait_queue(cfg, pres, starts=True, timeout=cfg.wait_start, with_windows=True)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kiauto/interposer.py", line 216, in wait_queue
    raise RuntimeError('Timed out waiting for `{}`'.format(strs))
RuntimeError: Timed out waiting for `['GTK:Window Title:', 'PANGO:0:']`
```

[run_erc_eeschema_screencast.ogv.zip](https://github.com/INTI-CMNB/KiAuto/files/14094229/run_erc_eeschema_screencast.ogv.zip)

set-soft, 2024-01-30:
Hi @LIV2 !

I didn't have time to check 7.0.10, the last working is 7.0.9

LIV2, 2024-01-30:
@set-soft with kicad 7.0.9 and interposer disabled it times out after
```
INFO:Waiting for "ERC File save dialog" ...
```
But works fine on 7.0.8

All versions time-out with the same timeout error as my previous comment when interposer is enabled.

VirtioFS works fine now though so that particular patch works

set-soft, 2024-01-30:
Hi @LIV2 !

Just to make things clear: Are you using the *KIAUTO_USE_PID_FOR_CREATE* environment variable?