INTI-CMNB/KiAuto #35: pcbnew_do  convert .pcbdoc Fail

chenzanyu, 2025-04-19:
The conversion automation operation was blocked by a pop-up.

[image]

chenzanyu, 2025-04-19:
Below are the details.   ```
# pcbnew_do -r convert  -o OUTPUT.kicad_pcb  test.PcbDoc ./
WARNING:KiCad project file not found (pcbnew_do.kiauto.file_util - file_util.py:317)
WARNING:Unknown KiCad dialog: KiCad PCB Editor Information (pcbnew_do - interposer.py:460)
WARNING:Potential dialog messages: {"Font 'Arial' not found; substituting 'Nimbus Sans Regular'.", "Font 'Calibri' not found; substituting 'DejaVu Sans Bold'.", "Font 'Arial' not found; substituting 'Nimbus Sans Bold'.", 'Details', '23:08:41', 'OK', 'H'} (pcbnew_do - interposer.py:461)
Traceback (most recent call last):
  File "/usr/bin/pcbnew_do", line 1947, in <module>
    id_pcbnew = wait_pcbnew_start_by_msg(cfg) if cfg.use_interposer else wait_pcbew_start(cfg)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/bin/pcbnew_do", line 1518, in wait_pcbnew_start_by_msg
    wait_start_by_msg(cfg)
  File "/usr/lib/python3/dist-packages/kiauto/interposer.py", line 723, in wait_start_by_msg
    res = wait_queue(cfg, pres, starts=True, timeout=cfg.wait_start, with_windows=True)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/kiauto/interposer.py", line 219, in wait_queue
    raise RuntimeError('Timed out waiting for `{}`'.format(strs))
RuntimeError: Timed out waiting for `['GTK:Window Title:', 'PANGO:0:']`
```

[test.zip](https://github.com/user-attachments/files/19823210/test.zip)

set-soft, 2025-04-21:
Hi @chenzanyu !
The above patch should solve your case, where layers has some mapping issues.
Can you check it? (if using docker images use the *dev* ones)

chenzanyu, 2025-04-21:
@set-soft  Thank you. Where can I obtain the development image? The latest image was released 7 days ago.

[image]

chenzanyu, 2025-04-22:
@set-soft  Thank you very much for your free answers! Using the latest development container for testing, test.PcbDoc can be converted normally. However, there is still an issue with the conversion of c.PcbDoc. Details are as follows:
```
pcbnew_do -r convert -o OUTPUT.kicad_pcb /project/input/c.PcbDoc /project/output/KicadProject
WARNING:KiCad project file not found (pcbnew_do.kiauto.file_util - file_util.py:317)
ERROR:Unknown KiCad dialog: KiCad PCB Editor Error (pcbnew_do - interposer.py:457)
ERROR:Potential dialog messages: {'Details', '20:13:40', 'H', "Font 'Arial Narrow' not found; substituting 'Nimbus Sans Narrow Regular'.", 'Missing null byte at end of property list. Imported data might be malformed or missing.', 'OK'} (pcbnew_do - interposer.py:458)
ERROR:Trace stack: (pcbnew_do - interposer.py:443)
  File "/usr/local/bin/pcbnew_do", line 1947, in <module>
    id_pcbnew = wait_pcbnew_start_by_msg(cfg) if cfg.use_interposer else wait_pcbew_start(cfg)
  File "/usr/local/bin/pcbnew_do", line 1518, in wait_pcbnew_start_by_msg
    wait_start_by_msg(cfg)
  File "/usr/local/lib/python3.11/dist-packages/kiauto/interposer.py", line 825, in wait_start_by_msg
    unknown_dialog(cfg, title)
```

This is the new Altium test file 

[c.zip](https://github.com/user-attachments/files/19856283/c.zip)

chenzanyu, 2025-04-22:
Also, please try    This file also failed to convert successfully as it was blocked by the pop-up window.  Details are as follows:
```
pcbnew_do -r convert -o OUTPUT.kicad_pcb /project/input/CM4-Main.PcbDoc /project/output/KicadProject
WARNING:KiCad project file not found (pcbnew_do.kiauto.file_util - file_util.py:317)
WARNING:Missing 'Tahoma' font (pcbnew_do - interposer.py:636)
WARNING:KiCad message: Font 'Tahoma' not found; substituting 'DejaVu Sans Book'. (pcbnew_do - interposer.py:637)
Traceback (most recent call last):
  File "/usr/local/bin/pcbnew_do", line 1947, in <module>
    id_pcbnew = wait_pcbnew_start_by_msg(cfg) if cfg.use_interposer else wait_pcbew_start(cfg)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/bin/pcbnew_do", line 1518, in wait_pcbnew_start_by_msg
    wait_start_by_msg(cfg)
  File "/usr/local/lib/python3.11/dist-packages/kiauto/interposer.py", line 742, in wait_start_by_msg
    res = wait_queue(cfg, pres, starts=True, timeout=cfg.wait_start, with_windows=True)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/kiauto/interposer.py", line 220, in wait_queue
    raise RuntimeError('Timed out waiting for `{}`'.format(strs))
RuntimeError: Timed out waiting for `['GTK:Window Title:', 'GTK:Window Set Modal:', 'PANGO:0:']`
```
Thank you again for your great work!

chenzanyu, 2025-04-25:
Hi @set-soft  In the latest container build, the above three Altium .PcbDoc files have been tested and are working correctly. Thank you very much for your prompt handling. I also found that pcbnew_do convert can be used to convert Eagle .brd files. It would be great if the eeschema_do command could also convert Altium and Eagle schematics (.SchDoc, .sch)

set-soft, 2025-04-25:
Please open a new issue as a "Feature Request" providing some example file. In this way I won't forget about it and I'll have some file for testing.

closed 2025-04-25 by set-soft