INTI-CMNB/KiAuto #36: I cannot get DRC to fail

rhodey, 2025-05-10:
I have written some software which automates creating PCB business cards

The input is a collection of text fields (name, company name, email, etc)

The output is a `.kicad_pcb` file which I then pass to `kibot` to generate gerbers using the built in Elecrow config / presets

I have had a number of these designs sent to Elecrow and they come back as they should but a few of my designs have been rejected due to DRC

I am trying to use this command to run DRC myself:

```
pcbnew_do run_drc pcb.kicad_pcb out/
```

I notice when I run this without a `pcb.kicad_pro` file in the dir the software auto-generates `pcb.kicad_pro`

I have taken the auto-generated `pcb.kicad_pro` and I have tried to modify DRC settings in a number of places in order to force a DRC fail

No matter how many configs I assign a huge number to I cant get this to fail, for example:

```
$ cat pcb.kicad_pro | grep 1000

        "copper_line_width": 1000.0,
        "fab_line_width": 1000.0,
        "other_line_width": 1000.0,
          "min_clearance": 1000.0
        "min_clearance": 1000.0,
        "min_connection": 1000.0,
        "min_track_width": 1000.0,
        "solder_mask_to_copper_clearance": 1000.0,
        "clearance": 1000.0,
```

The `pcb.kicad_pcb` file which is being used as input should fail DRC with these values being so large but always I get exit code `0`

And the file written to `out/drc_result.rpt` it looks like this:

```
** Drc report for pcb.kicad_pcb **
** Created on 2025-05-10T17:33:58+0000 **

** Found 0 DRC violations **

** Found 0 unconnected pads **

** Found 0 Footprint errors **

** End of Report **
```

The `pcbnew_do` program I am running from within a Docker container that starts with

```
FROM ghcr.io/inti-cmnb/kicad8_auto:1.7.0
```

I have added the input files pcb.kicad_pcb and pcb.kicad_pro to a zip archive you can find [here](https://rhodey.org/assets/pcb.zip)

One more thing I am noticing which seems like it could be related:

Every time I run `pcbnew_do run_drc` a file gets created in the directory, the file name is `pcb.kicad_pro-bak`

When I look inside this file I see that the majority is the same as `pcb.kicad_pro` but many of the settings I have changed to `1000.0` these settings are still default in the `-bak` file

```
$ cat pcb.kicad_pro | grep 1000
        "copper_line_width": 1000.0,
        "fab_line_width": 1000.0,
        "other_line_width": 1000.0,
          "min_clearance": 1000.0
        "min_clearance": 1000.0,
        "min_connection": 1000.0,
        "min_track_width": 1000.0,
        "solder_mask_to_copper_clearance": 1000.0,
        "clearance": 1000.0,
```

```
$ cat pcb.kicad_pro-bak | grep 1000
        "clearance": 1000.0,
```

Thank you for reading and please let me know whatever else more I can do to help, thanks!

rhodey, 2025-05-10:
I copied the contents of pcb.kicad_pro to a GitHub gist

https://gist.github.com/rhodey/fcfcc03dd958414f223b6d8ed345e725

This file is also found in the zip

set-soft, 2025-05-12:
> I notice when I run this without a `pcb.kicad_pro` file in the dir the software auto-generates `pcb.kicad_pro`

A lot of settings are stored in the project, so KiCad will create a project in a lot of situations.

> I have taken the auto-generated `pcb.kicad_pro` and I have tried to modify DRC settings in a number of places in order to force a DRC fail

Did you make it using the GUI?

> No matter how many configs I assign a huge number to I cant get this to fail, for example:
>
> ```
> $ cat pcb.kicad_pro | grep 1000
>
>         "copper_line_width": 1000.0,
>         "fab_line_width": [...]

You are setting random values, and using out of range numbers. 
You should open KiCad and set some real values.

> 
> The `pcb.kicad_pcb` file which is being used as input should fail DRC with these values being so large but always I get exit code `0`

If you run KiCad you'll see it reports 0 DRC errors.
Looking at your "board" I'm not sure if you'll be able to easily catch the issues.
KiCad doesn't seem to have a "copper to copper" distance for unconnected stuff.
What is even worst, you are defining whole layers as one footprint.

 
> And the file written to `out/drc_result.rpt` it looks like this:
> 
> ```
> ** Drc report for pcb.kicad_pcb **
> ** Created on 2025-05-10T17:33:58+0000 **
> 
> ** Found 0 DRC violations **
> 
> ** Found 0 unconnected pads **
> 
> ** Found 0 Footprint errors **
> 
> ** End of Report **
> ```

Which is the same you'll get running KiCad.
 

> The `pcbnew_do` program I am running from within a Docker container that starts with
> 
> ```
> FROM ghcr.io/inti-cmnb/kicad8_auto:1.7.0
> ```

Quite old, but not related to the problem.

> I have added the input files pcb.kicad_pcb and pcb.kicad_pro to a zip archive you can find [here](https://rhodey.org/assets/pcb.zip)
>
> One more thing I am noticing which seems like it could be [...]

This is a back-up

> When I look inside this file I see that the majority is the same as `pcb.kicad_pro` but many of the settings I have changed to `1000.0` these settings are still default in the `-bak` file

Yes, all the crazy values are reverted by KiCad itself.

 
> ```
> $ cat pcb.kicad_pro | grep 1000
>         "copper_line_width": 1000.0,
>         "fab_line_width": 1000.0,
>         "other_line_width": 1000.0,
>           "min_clearance": 1000.0 [...]

The only one that could affect DRC is this one.
But as you have one copper element in each layer it isn't a violation from KiCad point of view. 
KiCad can check distances between objects, not inside them.
The manufacturer is worried about the distance between copper polygons.

Ask in the KiCad forum is there are some mechanism, but KiCad DRC won't report violations in distances when both copper polygons belong to the same net. And you don't even have nets.

rhodey, 2025-05-17:
Hey thanks for the quick reply and sorry for my delay

It seems like you understand exactly what I am trying to do and why the DRC did not behave as I expected

I will write to the KiCad forum and see where that goes and I'll check back here with any updates

Thanks again!